letmaik/rawpy #121: RawPy fails to build on macOS

olokobayusuf, 2020-09-13:
In a few simple steps:
1. Run `git clone https://github.com/letmaik/rawpy.git`
2. `cd rawpy`
3. `pip install . --versbose`

The following errors pop up:
```
Scanning dependencies of target raw
  [  7%] Building CXX object CMakeFiles/raw.dir/private/var/folders/57/yczz_lsd1f94cd_j3nxm_ns00000gn/T/pip-req-build-cdckiqu4/external/LibRaw/internal/dcraw_common.cpp.o
  In file included from /private/var/folders/57/yczz_lsd1f94cd_j3nxm_ns00000gn/T/pip-req-build-cdckiqu4/external/LibRaw/internal/dcraw_common.cpp:19:
  In file included from /private/var/folders/57/yczz_lsd1f94cd_j3nxm_ns00000gn/T/pip-req-build-cdckiqu4/external/LibRaw/internal/dcraw_defs.h:22:
  In file included from /usr/local/include/libraw/libraw.h:377:
  /private/var/folders/57/yczz_lsd1f94cd_j3nxm_ns00000gn/T/pip-req-build-cdckiqu4/external/LibRaw/internal/libraw_internal_funcs.h:32:12: error: unknown type name 'libraw_static_table_t'
      static libraw_static_table_t tagtype_dataunit_bytes;
             ^
  /private/var/folders/57/yczz_lsd1f94cd_j3nxm_ns00000gn/T/pip-req-build-cdckiqu4/external/LibRaw/internal/libraw_internal_funcs.h:33:12: error: unknown type name 'libraw_static_table_t'
      static libraw_static_table_t Canon_wbi2std;
             ^
  /private/var/folders/57/yczz_lsd1f94cd_j3nxm_ns00000gn/T/pip-req-build-cdckiqu4/external/LibRaw/internal/libraw_internal_funcs.h:34:12: error: unknown type name 'libraw_static_table_t'
      static libraw_static_table_t Canon_KeyIsZero_Len2048_linenums_2_StdWBi;
             ^
  /private/var/folders/57/yczz_lsd1f94cd_j3nxm_ns00000gn/T/pip-req-build-cdckiqu4/external/LibRaw/internal/libraw_internal_funcs.h:35:12: error: unknown type name 'libraw_static_table_t'
      static libraw_static_table_t Canon_KeyIs0x0410_Len3072_linenums_2_StdWBi;
             ^
  /private/var/folders/57/yczz_lsd1f94cd_j3nxm_ns00000gn/T/pip-req-build-cdckiqu4/external/LibRaw/internal/libraw_internal_funcs.h:36:12: error: unknown type name 'libraw_static_table_t'
      static libraw_static_table_t Canon_KeyIs0x0410_Len2048_linenums_2_StdWBi;
             ^
  /private/var/folders/57/yczz_lsd1f94cd_j3nxm_ns00000gn/T/pip-req-build-cdckiqu4/external/LibRaw/internal/libraw_internal_funcs.h:37:12: error: unknown type name 'libraw_static_table_t'
      static libraw_static_table_t Canon_D30_linenums_2_StdWBi;
             ^
  /private/var/folders/57/yczz_lsd1f94cd_j3nxm_ns00000gn/T/pip-req-build-cdckiqu4/external/LibRaw/internal/libraw_internal_funcs.h:38:12: error: unknown type name 'libraw_static_table_t'
      static libraw_static_table_t Canon_G9_linenums_2_StdWBi;
             ^
  /private/var/folders/57/yczz_lsd1f94cd_j3nxm_ns00000gn/T/pip-req-build-cdckiqu4/external/LibRaw/internal/libraw_internal_funcs.h:40:12: error: unknown type name 'libraw_static_table_t'
      static libraw_static_table_t Fuji_wb_list1;
             ^
  /private/var/folders/57/yczz_lsd1f94cd_j3nxm_ns00000gn/T/pip-req-build-cdckiqu4/external/LibRaw/internal/libraw_internal_funcs.h:41:12: error: unknown type name 'libraw_static_table_t'
      static libraw_static_table_t FujiCCT_K;
             ^
  /private/var/folders/57/yczz_lsd1f94cd_j3nxm_ns00000gn/T/pip-req-build-cdckiqu4/external/LibRaw/internal/libraw_internal_funcs.h:42:12: error: unknown type name 'libraw_static_table_t'
      static libraw_static_table_t Fuji_wb_list2;
             ^
  /private/var/folders/57/yczz_lsd1f94cd_j3nxm_ns00000gn/T/pip-req-build-cdckiqu4/external/LibRaw/internal/libraw_internal_funcs.h:44:12: error: unknown type name 'libraw_static_table_t'
      static libraw_static_table_t Pentax_wb_list1;
             ^
  /private/var/folders/57/yczz_lsd1f94cd_j3nxm_ns00000gn/T/pip-req-build-cdckiqu4/external/LibRaw/internal/libraw_internal_funcs.h:45:12: error: unknown type name 'libraw_static_table_t'
      static libraw_static_table_t Pentax_wb_list2;
             ^
  /private/var/folders/57/yczz_lsd1f94cd_j3nxm_ns00000gn/T/pip-req-build-cdckiqu4/external/LibRaw/internal/libraw_internal_funcs.h:47:12: error: unknown type name 'libraw_static_table_t'
      static libraw_static_table_t Oly_wb_list1;
             ^
  /private/var/folders/57/yczz_lsd1f94cd_j3nxm_ns00000gn/T/pip-req-build-cdckiqu4/external/LibRaw/internal/libraw_internal_funcs.h:48:12: error: unknown type name 'libraw_static_table_t'
      static libraw_static_table_t Oly_wb_list2;
             ^
  /private/var/folders/57/yczz_lsd1f94cd_j3nxm_ns00000gn/T/pip-req-build-cdckiqu4/external/LibRaw/internal/libraw_internal_funcs.h:50:12: error: unknown type name 'libraw_static_table_t'
      static libraw_static_table_t Sony_SRF_wb_list;
             ^
  /private/var/folders/57/yczz_lsd1f94cd_j3nxm_ns00000gn/T/pip-req-build-cdckiqu4/external/LibRaw/internal/libraw_internal_funcs.h:51:12: error: unknown type name 'libraw_static_table_t'
      static libraw_static_table_t Sony_SR2_wb_list;
             ^
  /private/var/folders/57/yczz_lsd1f94cd_j3nxm_ns00000gn/T/pip-req-build-cdckiqu4/external/LibRaw/internal/libraw_internal_funcs.h:52:12: error: unknown type name 'libraw_static_table_t'
      static libraw_static_table_t Sony_SR2_wb_list1;
             ^
  /private/var/folders/57/yczz_lsd1f94cd_j3nxm_ns00000gn/T/pip-req-build-cdckiqu4/external/LibRaw/internal/libraw_internal_funcs.h:343:77: error: use of undeclared identifier 'LIBRAW_AHD_TILE'
      void ahd_interpolate_green_h_and_v(int top, int left, ushort (*out_rgb)[LIBRAW_AHD_TILE][LIBRAW_AHD_TILE][3]);
                                                                              ^
  /private/var/folders/57/yczz_lsd1f94cd_j3nxm_ns00000gn/T/pip-req-build-cdckiqu4/external/LibRaw/internal/libraw_internal_funcs.h:343:94: error: use of undeclared identifier 'LIBRAW_AHD_TILE'
      void ahd_interpolate_green_h_and_v(int top, int left, ushort (*out_rgb)[LIBRAW_AHD_TILE][LIBRAW_AHD_TILE][3]);
                                                                                               ^
  fatal error: too many errors emitted, stopping now [-ferror-limit=]
``` 

Any ideas?

letmaik, 2020-09-13:
Any specific reason why you're trying to build from source?

olokobayusuf, 2020-09-13:
Hey Maik, I need to expose the `libraw_colordata_t::linear_max` [member](https://www.libraw.org/docs/API-datastruct.html), as it might be the fix to the issue I opened earlier this year about pink skies (#100). I raised the issue with LibRaw, and [Alex mentioned that this field](https://www.libraw.org/node/2605) could be the way to compute the proper saturation level that fixes the hot pink that comes up in Canon RAW's.

letmaik, 2020-09-13:
I don't have a mac myself so the only help I can offer is to point you to the CI script that does the macOS builds: https://github.com/letmaik/rawpy/blob/master/.github/scripts/build-macos.sh It's certainly not as straight-forward as for Linux.

letmaik, 2020-09-13:
An alternative is that you create a fork and commit your changes to a branch. Then GitHub Actions will automatically build it and you can download the wheel from the build artifacts.

olokobayusuf, 2020-09-13:
Would you consider exposing the `linear_max` in the `RawPy` class? That way once I fork and if we can get it to work, you can add it to the next release (I can create a PR from my fork). Gonna fork it and make the mod now.

olokobayusuf, 2020-09-13:
@letmaik that worked. I've opened #122. I'm not sure how you schedule releases, but what do you think about merging this?

letmaik, 2020-09-13:
Sure, I don't see a problem in that. If I understand correctly, then `linear_max` is essentially the camera-provided per-channel white level, right? If that's correct, I would expose it as `camera_white_level_per_channel`.

olokobayusuf, 2020-09-13:
According to the docs, that's correct. And for extra parity with `black_level_per_channel`, I think you'd prefer `white_level_per_channel` instead of `camera_white_level_per_channel` ;). Lmk and I'll rename and push.

Thank you so much by the way. With this, I'll go ahead and close #100. Also, do you mind if I reach out for intros? I think I sent you an invitation on LinkedIn a while ago. I'm cofounder at a computational photography research startup, @homedeck. I'd love to keep in touch.

letmaik, 2020-09-13:
I think there is a difference in semantics though. `black_level_per_channel` I think may be computed from data, whereas `linear_max` is always just the metadata and not based on the actual data. That's why I was going for the added `camera_` prefix to make it clearer. I guess the docstring for `white_level` could be extended. The official docs say "Maximum pixel value. Calculated from the data for most cameras, hardcoded for others."

Regarding release schedule, there is no schedule :) If someone needs something urgent I'm happy to do a release, instead of collecting multiple bug fixes.

olokobayusuf, 2020-09-13:
That's fair. I've updated the docstring on `white_level` and renamed the linear max property to `camera_white_level_per_channel`. Feel free to merge when the build is finished.

closed 2020-12-17 by letmaik