letmaik/rawpy #126: unable to open 4224 x 3136 raw image

peteasa, 2020-10-28:
I have used ImageJ to open this 4224 x 3136 little-endian byte order file but for some reason rawpy complains:
`    with rawpy.imread('template.raw') as raw:
  File "/home/pas/.local/lib/python3.6/site-packages/rawpy/__init__.py", line 20, in imread
    d.open_file(pathOrFile)
  File "rawpy/_rawpy.pyx", line 404, in rawpy._rawpy.RawPy.open_file
  File "rawpy/_rawpy.pyx", line 914, in rawpy._rawpy.RawPy.handle_error
rawpy._rawpy.LibRawFileUnsupportedError: b'Unsupported file format or not RAW file'
`
Is there a way to get additional debug or a way to fix this image loading problem?

`
 //  Captured 13:52:29 - Wednesday, October 28, 2020
//
//  Frame 10 of 10
//  Sensor info:
//	Width                 = 4224
//	Height                = 3136
//	Top Non-Pixel Rows    = 0
//	Bottom Non-Pixel Rows = 0
//	Image Format          = Bayer 10
//	Subformat             = GRBG
//	Sensor output         = 10 bits per pixel
//	Color Filter          = Monochrome
//
//  .RAW file info:
//	stored as      = 16 bits unsigned short
//	Endianess      = little
//             bit|  7  |  6  |  5  |  4  |  3  |  2  |  1  |  0  |
//        Byte 0  | ------------------Pixel[7:0]----------------- |
//        Byte 1  |  0     0     0     0     0     0  | Pixel[9:8]|
//
//  Application info:
//	Application Name = C:/Aptina Imaging/bin/DevWareX.exe
//	Application Version = 6.0.25.54771
//	Application Type = 6.0.25_Release
//	Application Date = 03/12/2020
`

peteasa, 2020-10-28:
image file is here
[img.zip](https://github.com/letmaik/rawpy/files/5452136/img.zip)

letmaik, 2020-10-28:
What camera is the file from?

peteasa, 2020-10-29:
ON Semiconductor - Sensor Part Number = AR1335

I found that I could read and write the file directly with numpy so I am not now using rawpy.. suspect I am loosing some file header information along the way!
``def read_write_raw():
    with open('template.raw', 'r') as infile:
        data = np.fromfile(infile, dtype=np.uint16)
        pixels, = data.shape
        height = int(3136)
        width = int(pixels / height)
        data = np.reshape(data, (height, width))
        im = Image.fromarray(data)
        im.save('template1.png')
        data.astype(np.uint16).tofile('template1.raw')
``

kmilos, 2020-10-30:
Yep, this is just a really raw bytestream dump straight out of the sensor (i.e. not a raw file format produced by more sophisticated camera software), so reading it in using numpy directly is about the only thing you can do*.

Your file is 26492928 bytes = 4224 x 3136 x 2 bytes, there literally isn't anything else in there.

rawpy/LibRaw can only parse actual "raw" file containers that also come with mandatory metadata helping them to interpret the bytes within.

* If a  proper raw file is absolutely necessary, you could wrap your raw bytestream with the metadata you know/record separately in e.g. a DNG file using either [libtiff](https://discuss.pixls.us/t/raspberry-pi-camera-sony-imx2019-8mp-raw-format/13412/5), or something along the lines of the [pgm2dng](https://www.libraw.org/node/2579) tool.

However, there also seems to be a [low-level open_bayer() method in LibRaw](https://www.libraw.org/node/2538), but I'm not sure if it makes sense for rawpy to expose that...

letmaik, 2020-10-30:
I agree with @kmilos that wrapping as DNG would make the most sense in order to load it up in rawpy and be able to demosaic it. The new `open_bayer()` function of LibRaw would theoretically allow to bypass this step but for that I would probably have to expose a lot more of LibRaw in order to set all required metadata options for it do to the right kind of processing. Given that this is a rather niche use case (most people just load up regular camera raw images) I'd rather not do it.

closed 2020-10-30 by letmaik

kmilos, 2020-10-30:
Btw, in the thread I mentioned, there was a [Python only attempt at DNG creation](https://discuss.pixls.us/t/raspberry-pi-camera-sony-imx2019-8mp-raw-format/13412/14) that seems promising. Needs some patching of pylibtiff though...

marcel001, 2021-01-26:
> I have used ImageJ to open this 4224 x 3136 little-endian byte order file but for some reason rawpy complains:
> ` with rawpy.imread('template.raw') as raw: File [...]

Hi peteasa! Did you have any luck with this issue? I also need to open a bytestream that is lacking any metadata/header. Did you find a solution? Thanks!

peteasa, 2021-01-26:
Hi,

In my case I just open the data as a numpy array.  All my images are the same size so it is easy for me

data = np.fromfile(infile, dtype=np.uint16)
data = np.reshape(data, (height, width))
data.tofile(outfile)

HongChow, 2022-03-04:
> Hi,
> 
> In my case I just open the data as a numpy array. All my images are the same size so it is easy for me
> 
> data = np.fromfile(infile, dtype=np.uint16) data = np.reshape(data, (height, width)) data.tofile(outfile)

hi , could you still use raw post process using numpy data array?