letmaik/rawpy #127: Unsupported file format or not RAW file

Sanket-15, 2020-11-02:
Hello,

I'm trying to open a '.raw' file using the following command : rawpy.imread(image_path).
I'm receiving the following error:  rawpy._rawpy.LibRawFileUnsupportedError: b'Unsupported file format or not RAW file'

Description about my '.raw' file:
The '.raw' file does not contain any header information. Each pixel is stored using 2 bytes/pixels.

Please help me out if anyone faced the same problem.

kmilos, 2020-11-04:
Duplicate of #126?

Sanket-15, 2020-11-05:
Hello @kmilos ,

Thank you for pointing me out on the right direction.

However I'm confused as to how to convert raw file to pgm file using [rawtopgm](http://netpbm.sourceforge.net/doc/rawtopgm.html).

Can you please help?

The Environment I'm using is Anaconda and Python 3

letmaik, 2020-12-17:
Closing, this question is better suited for stackoverflow.

closed 2020-12-17 by letmaik