letmaik/rawpy #130: Edit attributes other than Rawpy.raw_image ?

amonod, 2020-12-22:
Hi, 

First I wanted to say thank you for making this library. It goes without saying that it is extremely convenient for many python image processing applications.

My actual question is the following: when a raw file is opened with 
`rawpy.imread(filePath)`, we can change the values of the Bayer array by affecting new values to the `raw_image` attribute of the `Rawpy` object. I've noticed, however, that we cannot edit the values of other array attributes, such as `Rawpy.black_level_per_channel`, or `Rawpy.color_matrix` . The potential usefulness of being able to edit such attributes is pretty obvious. Why is it not possible?

letmaik, 2020-12-22:
These are related to postprocessing. You can specify them as arguments in the `.postprocess(..)` method. See also the API docs.

closed 2020-12-22 by letmaik

amonod, 2020-12-22:
Thank you, I had not seen the `user_black` parameter of postprocessing. However, there does not seem to be an option to change the color matrix.

letmaik, 2020-12-22:
These are all the supported options and how they are passed down to libraw: https://github.com/letmaik/rawpy/blob/5e510fbc5f7b75f99380bd71d83b3d3a7db06888/rawpy/_rawpy.pyx#L893-L928
If a libraw option (see https://www.libraw.org/docs/API-datastruct-eng.html#libraw_output_params_t) is missing that you need, please open an issue specifically for that.