letmaik/rawpy #143: The shapes of `raw_colors` and `raw_image` are different.

manipopopo, 2021-05-24:
For the [Fuji X-Pro 1 raw](http://www.rawsamples.ch/raws/fuji/RAW_FUJI_XPRO1.RAF) example, the shape of `raw_colors` is `(3288, 5118)`, and the shape of `raw_image` is `(3288, 5120)`.

Maybe the `raw_colors` could be created by
```python
np.pad(pattern, ((0, max(height - n, 0)), (0, max(width - n, 0))), mode='wrap')
```
instead of 
https://github.com/letmaik/rawpy/blob/ae822649e72ad112d9aa86a8001fa5cbe708d7a4/rawpy/_rawpy.pyx#L615

letmaik, 2021-05-24:
This is the output from exiftool:
> Raw Image Full Size             : 5120x3288
Raw Image Crop Top Left         : 12 28
Raw Image Cropped Size          : 4896x3264
Fuji Layout                     : 12 12 12 12
X-Trans Layout                  : GRBGBR BGGRGG RGGBGG GBRGRB RGGBGG BGGRGG

And `raw_pattern` from rawpy:
```
>>> r.raw_pattern
array([[1, 1, 0, 1, 1, 2],
       [1, 1, 2, 1, 1, 0],
       [2, 0, 1, 0, 2, 1],
       [1, 1, 2, 1, 1, 0],
       [1, 1, 0, 1, 1, 2],
       [0, 2, 1, 2, 0, 1]], dtype=uint8)
```

We could of course just pad to the correct size as you suggest but I'm wondering what the underlying issue is and whether that's even the correct solution.

letmaik, 2021-05-24:
I opened #144. It's probably all fine. I guess there's no reason why the raw size would have to be a multiple of the Bayer pattern size.

PR #144 (2021-05-25): Fix raw_colors for fuji xtrans
Fixes #143
closed 2021-05-25 by letmaik