letmaik/rawpy #145: pipenv installation is broken

stunpix, 2021-06-05:
Hello,

I was trying to install rawpy using `pipenv install rawpy` and got this error:

```
An error occurred while installing rawpy==0.13.1 --hash=sha256:5e723097fa5e1bc8b3b97b6813be6a33d2476d2264e5111881820349f4a44f4d --hash=sha256:9d5ed61efaf55b8b18a37ed0a6e322c3747fa897457ccad1a0f764e9808b6bd1 --hash=sha256:4af5ef88cd11568a16ce29d30bd69cecc290d14bf32cf57917a095ac3138dd0c --hash=sha256:ef5e860e45cf172a877d26033f744165ceb971c7e87bdce75b912d94d1e5ad7d --hash=sha256:38ec2e79e61bda67d99f33843d7647d6b344aa82e79e5aa85d189d2242891de3 --hash=sha256:962878ee931b0a2da95723334d09fd621bcc7495477eda5e380d906619adb1a0 --hash=sha256:7aaffe1867893586ffe71293c6315092603de606365aa5e5946abd4190ef7851 --hash=sha256:e2643d3c9c486e49bac1a2465179745ba7afa28edf5ea73358aabe97ca739e0d --hash=sha256:35f5c69cc5598b193e2b5c1d6519e8d45cb2cc9cbacc8c4c636af6db00db6e44 --hash=sha256:8dcbed5089258d80067e6342018cbecf3b9a75ab4fe19c2096ee4f0c7141955a --hash=sha256:39a4ed419c3aea2a6829afba0f2cd0559b45822343a9b72d93895d65fbadbfd5 --hash=sha256:ee2d310abf9c88c564c8a7ac34409f8dc3e3cd2c0439385a0745c87a1f13d84b --hash=sha256:2f3a67d1836be46874de138f75ebdfc6fa9abcd166d04ad0990706862a9531db --hash=sha256:207d2fbbfb30166ca56123b1e17726207bec068ee26cd50fedb319883f5442d2 --hash=sha256:a13ec89b7bf598f51e16bb85318ea72a2b6f566b33b7b084e059d4ade469c5c8 --hash=sha256:ae17018bef7869df24d8bcb0ec0a839b035d96f1a9501ca7bc986ffcb56cb7a6 --hash=sha256:771550565e2bf6cf1955e059f2ed6f03b555997c8b3bbedfd654095002093235 --hash=sha256:8f83fa1e73e7a70ba3d1a9e84f11908b5163786624bbde7acfdcadc99a1e9783 --hash=sha256:e5b2e29f03672b8debc261c3874feb63d78cc961cc791fb685025ba1de021d3f --hash=sha256:4b3fc84acaab2a7ff20605988618aec830a09665cba2bc8d11b9227c1fa18e08 --hash=sha256:d78a0035a3e5f6f03c2080698806f06ae7837f27bf325631cfe3e05b9c3f7dfa! Will try again.
Installing initially failed dependencies...
[InstallError]:   File "/Users/olks/Library/Python/3.8/lib/python/site-packages/pipenv/cli/command.py", line 233, in install
[InstallError]:       retcode = do_install(
[InstallError]:   File "/Users/olks/Library/Python/3.8/lib/python/site-packages/pipenv/core.py", line 2193, in do_install
[InstallError]:       do_init(
[InstallError]:   File "/Users/olks/Library/Python/3.8/lib/python/site-packages/pipenv/core.py", line 1304, in do_init
[InstallError]:       do_install_dependencies(
[InstallError]:   File "/Users/olks/Library/Python/3.8/lib/python/site-packages/pipenv/core.py", line 899, in do_install_dependencies
[InstallError]:       batch_install(
[InstallError]:   File "/Users/olks/Library/Python/3.8/lib/python/site-packages/pipenv/core.py", line 796, in batch_install
[InstallError]:       _cleanup_procs(procs, failed_deps_queue, retry=retry)
[InstallError]:   File "/Users/olks/Library/Python/3.8/lib/python/site-packages/pipenv/core.py", line 703, in _cleanup_procs
[InstallError]:       raise exceptions.InstallError(c.dep.name, extra=err_lines)
[pipenv.exceptions.InstallError]: Collecting rawpy==0.13.1
[pipenv.exceptions.InstallError]:   Using cached rawpy-0.13.1.tar.gz (116 kB)
[pipenv.exceptions.InstallError]: Building wheels for collected packages: rawpy
[pipenv.exceptions.InstallError]:   Building wheel for rawpy (setup.py): started
[pipenv.exceptions.InstallError]:   Building wheel for rawpy (setup.py): finished with status 'error'
[pipenv.exceptions.InstallError]:   ERROR: Command errored out with exit status 1:
[pipenv.exceptions.InstallError]:    command: /Users/olks/.local/share/virtualenvs/high-dynamic-range-image-bU1bBh3H/bin/python -u -c 'import sys, setuptools, tokenize; sys.argv[0] = '"'"'/private/var/folders/pv/ww8wpvvj73l844c56kvhs7yw0000gn/T/pip-install-cvtqg8yw/rawpy_3c5ff783a98f44b0b21dee65747119ba/setup.py'"'"'; __file__='"'"'/private/var/folders/pv/ww8wpvvj73l844c56kvhs7yw0000gn/T/pip-install-cvtqg8yw/rawpy_3c5ff783a98f44b0b21dee65747119ba/setup.py'"'"';f=getattr(tokenize, '"'"'open'"'"', open)(__file__);code=f.read().replace('"'"'\r\n'"'"', '"'"'\n'"'"');f.close();exec(compile(code, __file__, '"'"'exec'"'"'))' bdist_wheel -d /private/var/folders/pv/ww8wpvvj73l844c56kvhs7yw0000gn/T/pip-wheel-8l75qbtd
[pipenv.exceptions.InstallError]:        cwd: /private/var/folders/pv/ww8wpvvj73l844c56kvhs7yw0000gn/T/pip-install-cvtqg8yw/rawpy_3c5ff783a98f44b0b21dee65747119ba/
[pipenv.exceptions.InstallError]:   Complete output (11 lines):
[pipenv.exceptions.InstallError]:   LibRaw git submodule is not cloned yet, will invoke "git submodule update --init" now
[pipenv.exceptions.InstallError]:   fatal: not a git repository (or any of the parent directories): .git
[pipenv.exceptions.InstallError]:   Traceback (most recent call last):
[pipenv.exceptions.InstallError]:     File "<string>", line 1, in <module>
[pipenv.exceptions.InstallError]:     File "/private/var/folders/pv/ww8wpvvj73l844c56kvhs7yw0000gn/T/pip-install-cvtqg8yw/rawpy_3c5ff783a98f44b0b21dee65747119ba/setup.py", line 298, in <module>
[pipenv.exceptions.InstallError]:       mac_libraw_compile()
[pipenv.exceptions.InstallError]:     File "/private/var/folders/pv/ww8wpvvj73l844c56kvhs7yw0000gn/T/pip-install-cvtqg8yw/rawpy_3c5ff783a98f44b0b21dee65747119ba/setup.py", line 259, in mac_libraw_compile
[pipenv.exceptions.InstallError]:       clone_submodules()
[pipenv.exceptions.InstallError]:     File "/private/var/folders/pv/ww8wpvvj73l844c56kvhs7yw0000gn/T/pip-install-cvtqg8yw/rawpy_3c5ff783a98f44b0b21dee65747119ba/setup.py", line 127, in clone_submodules
[pipenv.exceptions.InstallError]:       raise Exception('git failed')
[pipenv.exceptions.InstallError]:   Exception: git failed
[pipenv.exceptions.InstallError]:   ----------------------------------------
[pipenv.exceptions.InstallError]:   ERROR: Failed building wheel for rawpy
[pipenv.exceptions.InstallError]:   Running setup.py clean for rawpy
[pipenv.exceptions.InstallError]: Failed to build rawpy
[pipenv.exceptions.InstallError]: Installing collected packages: rawpy
[pipenv.exceptions.InstallError]:   Attempting uninstall: rawpy
[pipenv.exceptions.InstallError]:     Found existing installation: rawpy 0.16.0
[pipenv.exceptions.InstallError]:     Uninstalling rawpy-0.16.0:
[pipenv.exceptions.InstallError]:       Successfully uninstalled rawpy-0.16.0
[pipenv.exceptions.InstallError]:     Running setup.py install for rawpy: started
[pipenv.exceptions.InstallError]:     Running setup.py install for rawpy: finished with status 'error'
[pipenv.exceptions.InstallError]:     ERROR: Command errored out with exit status 1:
[pipenv.exceptions.InstallError]:      command: /Users/olks/.local/share/virtualenvs/high-dynamic-range-image-bU1bBh3H/bin/python -u -c 'import sys, setuptools, tokenize; sys.argv[0] = '"'"'/private/var/folders/pv/ww8wpvvj73l844c56kvhs7yw0000gn/T/pip-install-cvtqg8yw/rawpy_3c5ff783a98f44b0b21dee65747119ba/setup.py'"'"'; __file__='"'"'/private/var/folders/pv/ww8wpvvj73l844c56kvhs7yw0000gn/T/pip-install-cvtqg8yw/rawpy_3c5ff783a98f44b0b21dee65747119ba/setup.py'"'"';f=getattr(tokenize, '"'"'open'"'"', open)(__file__);code=f.read().replace('"'"'\r\n'"'"', '"'"'\n'"'"');f.close();exec(compile(code, __file__, '"'"'exec'"'"'))' install --record /private/var/folders/pv/ww8wpvvj73l844c56kvhs7yw0000gn/T/pip-record-uk44rxiq/install-record.txt --single-version-externally-managed --compile --install-headers /Users/olks/.local/share/virtualenvs/high-dynamic-range-image-bU1bBh3H/include/site/python3.9/rawpy
[pipenv.exceptions.InstallError]:          cwd: /private/var/folders/pv/ww8wpvvj73l844c56kvhs7yw0000gn/T/pip-install-cvtqg8yw/rawpy_3c5ff783a98f44b0b21dee65747119ba/
[pipenv.exceptions.InstallError]:     Complete output (11 lines):
[pipenv.exceptions.InstallError]:     LibRaw git submodule is not cloned yet, will invoke "git submodule update --init" now
[pipenv.exceptions.InstallError]:     fatal: not a git repository (or any of the parent directories): .git
[pipenv.exceptions.InstallError]:     Traceback (most recent call last):
[pipenv.exceptions.InstallError]:       File "<string>", line 1, in <module>
[pipenv.exceptions.InstallError]:       File "/private/var/folders/pv/ww8wpvvj73l844c56kvhs7yw0000gn/T/pip-install-cvtqg8yw/rawpy_3c5ff783a98f44b0b21dee65747119ba/setup.py", line 298, in <module>
[pipenv.exceptions.InstallError]:         mac_libraw_compile()
[pipenv.exceptions.InstallError]:       File "/private/var/folders/pv/ww8wpvvj73l844c56kvhs7yw0000gn/T/pip-install-cvtqg8yw/rawpy_3c5ff783a98f44b0b21dee65747119ba/setup.py", line 259, in mac_libraw_compile
[... cut ...] mpile --install-headers /Users/olks/.local/share/virtualenvs/high-dynamic-range-image-bU1bBh3H/include/site/python3.9/rawpy Check the logs for full command output.
ERROR: Couldn't install package: rawpy
```
I guess pipenv runs `setup.py` with some option and it tries to install everything from source due to [this evil hack](https://github.com/letmaik/rawpy/blob/935953ef05e0aa50c460fb607f303afb94ec42a6/setup.py#L267) rather than use wheels.

My environment: macOS 11.4 and python 3.9 from macports

letmaik, 2021-06-05:
Two questions:
- Why are you using an outdated rawpy version (0.13.1)?
- Why aren't you using the binary wheels from pypi?

In general I don't support building from source, which is why I don't publish the sdist anymore to PyPI. It's too error-prone. If you really want to build from source then please consult the README, but you're on your own then.

stunpix, 2021-06-05:
Sorry, seems I’ve been not clear enough. I’m not trying to install from sources. I'm only trying to use pipenv in a usual way even without specifying version. I have no clue why pipenv or underlaying pip had selected outdated version. I’ll try to manage out how to install latest version perhaps it has no such issues

letmaik, 2021-06-08:
I'm going to close this since it's not an issue with rawpy itself. If you figure out what the problem is please feel free to post your solution here.

closed 2021-06-08 by letmaik

mgolub2, 2021-06-16:
@stunpix I believe I know what happened here - Are you on an M1 Mac? I had this same issue just from within a virtualenv, and it's because there's no pre-built wheel for arm. An easy workaround is to use rosetta2+x86 python then pipenv/pip worked fine for me without issues.

stunpix, 2021-06-18:
@mgolub2 that's not the case for me since I'm using intel macbook 16" not M1. I decided to not struggle with this issue since pip is working perfectly and I dunno what's wrong with pipenv and why it's selecting the old version of rawpy. I totally cleaned up and reinstalled python environment and still getting the same error.