letmaik/rawpy #159: Add support to release linux aarch64 wheels

odidev, 2021-11-12:
Problem
---------
On aarch64, ‘pip install rawpy’ is giving the below error - 
```
ERROR: Could not find a version that satisfies the requirement rawpy (from versions: none) 
ERROR: No matching distribution found for rawpy 
```
Resolution
-----------
On aarch64, ‘pip install rawpy’ should download the wheels from pypi 

@letmaik and Team Please let me know your interest in releasing aarch64 wheels. I can help in this.

letmaik, 2021-11-12:
Out of curiosity, which device are you using that would benefit from such wheels?

odidev, 2021-11-15:
It will be useful if the aarch64 wheel is released as - 

- The Arm architecture has been available on AWS (Graviton2) for about 2 years now and has seen a lot of adoption by various organizations. The install base of Arm users is growing at a rapid pace. 

- Oracle cloud will be releasing Arm based Compute Shapes in the future. Please [see](https://www.anandtech.com/show/16100/oracle-announces-upcoming-cloud-compute-instances-ice-lake-and-milan-a100-and-altra#:~:text=OCI%E2%80%99s%20Arm%20instances%20will%20clock%20as%20high%20as,%E2%80%93%20OCI%20is%20refreshing%20its%20GPU%20compute%20instances)

- Generally, there is a lot of momentum behind various products switching to Arm based CPUs. For example, the Microsoft surface Pro X is an Arm based device.

letmaik, 2021-11-21:
Closing this since https://github.com/letmaik/rawpy/pull/160 is merged and aarch64 wheels will be part of the next release.

closed 2021-11-21 by letmaik