letmaik/rawpy #167: Rawpy pip package wrong architecture on M1 Mac (arm64)

programatt, 2022-01-28:
I have installed libraw with `brew install libraw` and rawpy with `pip install rawpy`. However I receive the following error when trying to import rawpy in my code

```
% python3
Python 3.10.1 (main, Dec  6 2021, 22:18:13) [Clang 13.0.0 (clang-1300.0.29.3)] on darwin
Type "help", "copyright", "credits" or "license" for more information.
>>> import rawpy
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "/Users/matt/code/venvs/tc/lib/python3.10/site-packages/rawpy/__init__.py", line 5, in <module>
    import rawpy._rawpy
ImportError: dlopen(/Users/matt/code/venvs/tc/lib/python3.10/site-packages/rawpy/_rawpy.cpython-310-darwin.so, 0x0002): tried: '/Users/matt/code/venvs/tc/lib/python3.10/site-packages/rawpy/_rawpy.cpython-310-darwin.so' (mach-o file, but is an incompatible architecture (have 'x86_64', need 'arm64e')), '/usr/local/lib/_rawpy.cpython-310-darwin.so' (no such file), '/usr/lib/_rawpy.cpython-310-darwin.so' (no such file)
```

I don't really understand how python packaging and wheels work but my understanding is the `--no-binary :all:` option should try and build the package from source but I got this error.

```
pip install rawpy --no-binary :all:
ERROR: Could not find a version that satisfies the requirement rawpy (from versions: none)
ERROR: No matching distribution found for rawpy
```

Does rawpy support running on arm64 systems?

letmaik, 2022-01-28:
rawpy doesn't publish arm64 wheels yet (because GitHub Actions doesn't have M1 runners yet), but for some reason the x64 wheel for Python 3.10 is tagged as universal2 which it shouldn't be. This is why it successfully installs for you but then fails during import.
You can build from source by following https://github.com/letmaik/rawpy#installation-from-source-on-linuxmacos. Note that arm64 hasn't been tested, so you're on your own here.
I'll leave this issue open to deal with the mistagged 3.10 wheel.

programatt, 2022-01-28:
@letmaik thanks, I was able to get it to install from source and it imports ok. I will try using it now. Do you want me to report any issues with arm64 that I run into on a different issue or this one?

letmaik, 2022-01-28:
For arm64 issues, please open new issues please.

Regarding the tagging issue, seems like this is known: https://github.com/pypa/setuptools/issues/2520#ref-pullrequest-807916179

closed 2022-02-24 by letmaik

kamikaze, 2022-12-27:
u don't need to wait for arm64 runners, use docker buildx actions, take a look: https://github.com/kamikaze/fastapi-auth-service/tree/master/.github/workflows

commit 12cf40db (2022-02-24): fix macos wheel tag for py 3.10
(fixes #167)