letmaik/rawpy #172: Does this "postprocess()" contain tonemapping?

TIMESTICKING, 2022-04-25:
Hi, i was using matlab to read a .dng file, and the bit depth is larger than 8 bits, so I directly divide them by the max of them. the result I got is lighter than yours. I set `postprocess(output_bps=8, use_camera_wb=False, gamma=(1,1))`
the matlab code is 
```
function [ryb,ExposureTime,ISO,AsShotNeutral] = dng2ryb(filename, isrotate)
%UNTITLED4 Summary of this function goes here
%   Detailed explanation goes here
warning off %MATLAB:tifflib:TIFFReadDirectory:libraryWarning
t = Tiff(filename);
cfa = read(t);
close(t);

cfa = max(0,cfa - 256);

ryb = zeros(size(cfa,1),size(cfa,2),3);

ryb(1:2:end,1:2:end,3) = double(cfa(1:2:end,1:2:end,:));
ryb(2:2:end,2:2:end,1) = double(cfa(2:2:end,2:2:end,:));

ryb(1:2:end,2:2:end,2) = double(cfa(1:2:end,2:2:end,:));
ryb(2:2:end,1:2:end,2) = double(cfa(2:2:end,1:2:end,:));

ryb(:,:,2) = ryb(:,:,2) + imfilter(ryb(:,:,2), [0 1 0; 1 0 1; 0 1 0]/4);
    
B1 = imfilter(ryb(:,:,3),[1 0 1; 0 0 0; 1 0 1]/4);
B2 = imfilter(ryb(:,:,3)+B1,[0 1 0; 1 0 1; 0 1 0]/4);
ryb(:,:,3) = ryb(:,:,3) + B1 + B2;
    
R1 = imfilter(ryb(:,:,1),[1 0 1; 0 0 0; 1 0 1]/4);
R2 = imfilter(ryb(:,:,1)+R1,[0 1 0; 1 0 1; 0 1 0]/4);
ryb(:,:,1) = ryb(:,:,1) + R1 + R2;
```
I know I got RYB from those, so after I've done this `rgb(:,:,2) = ryb(:,:,2) - ryb(:,:,1);` 
So, I just want to get the original the raw picture, does your lib do something else? Like tone-mapping, gamma trans ...

kmilos, 2022-04-26:
In addition to setting gamma to 1, you also might want to avoid color space conversion (to sRGB by default) by setting [`output_color=ColorSpace.raw`](https://letmaik.github.io/rawpy/api/rawpy.Params.html). I usually do `user_wb=(1,1,1,1), output_color=ColorSpace.raw, output_bps=16, gamma=(1,1)` to get something as close to raw as possible (this will still include black level subtract and stretching/scaling to maximum data range).

Please use [Discussions](https://github.com/letmaik/rawpy/discussions) for questions like these, this section is for reporting real issues.

TIMESTICKING, 2022-04-26:
thank you a lot! :D

closed 2022-04-26 by TIMESTICKING