letmaik/rawpy #176: how to set user_black value for each channel separately?

junyongjeon, 2022-07-01:
To crop noise in low grayscale regions, I want to use 'user_black' to crop very low grayscale regions, but since I can only set one int value for 'user_black', I can't subtract a separate value for each color. As a result, it gives the image an awkward color.
Is there any method to provide separate 'user_black' for each color channel?

letmaik, 2022-07-11:
libraw (which rawpy is built on) does support per-channel black level corrections via `user_cblack[4]`. So in principle this could be exposed as a new parameter in rawpy.
Could you give an example of how you would determine the per-channel correction values that you would use in this case?

sydney-d, 2025-01-17:
Hello,

Have this been implemented ? 
Could be useful to correct colors on an inverted color film negative, when white balancing on the base is not enough.

Thanks :)