letmaik/rawpy #178: ImportError: DLL load failed while importing _rawpy:

yeahitsjan, 2022-07-09:
First of all! Awesome library. I really like it and would test out some things on Windows. So I installed a fresh version of Windows 10, installed Python 3.10.5 and Visual Studio Code (the machine is empty). With ``pip install rawpy`` I installed rawpy and trying to import it gives me this result:

```
>>> import rawpy
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "C:\Python310\lib\site-packages\rawpy\__init__.py", line 5, in <module>
    import rawpy._rawpy
ImportError: DLL load failed while importing _rawpy: The specified module was not found.
```

What can I do about it? Do I miss some dependencies?

Many thanks,
Jan

letmaik, 2022-07-10:
That shouldn't happen... I'll look into it, but in the meantime you can fix the issue by installing the latest Microsoft Visual C++ redistributable from https://docs.microsoft.com/en-US/cpp/windows/latest-supported-vc-redist

yeahitsjan, 2022-07-10:
Your solution worked fine. So the Redist pack is a requirement on a fresh Windows install.

letmaik, 2022-07-10:
Thanks for reporting back. For reference, this is the same as described at https://github.com/pypa/cibuildwheel/blob/main/docs/faq.md#windows-importerror-dll-load-failed-the-specific-module-could-not-be-found. According to this page, the latest Python versions should come bundled with the required Visual C++ redistributable, so I'm not really sure why the issue happens anyway. I'm going to close the issue since a reasonable work-around exists and because the issue likely only appears on fresh/unpatched Windows installations.

closed 2022-07-10 by letmaik