letmaik/rawpy #179: How to extract bayer pattern

thomaskleiven, 2022-08-08:
Is it possible to split the raw data up into its red, green, and blue components _before_ demosaicing? 

It seems like that raw data is available using `raw_image` ([docs](https://letmaik.github.io/rawpy/api/rawpy.RawPy.html#rawpy.RawPy.raw_image))
```python
file_path = 'sample1.dng'
with rawpy.imread(file_path) as raw:
    img = raw.raw_image   

    print(img[0,0]) ## Prints 158
```

Provided that the sensor data is on the format RGGB, then `img[0,0]` would correspond to the first red pixel of the sensor data. 

___________________________________

**Goal**: Extract the value of the first red pixel

Method 1)
Using `rawpy`, the value of `img[0,0]` is `158`. 

Method 2)
Using `RawTherapee` ([docs](https://www.rawtherapee.com)) to read the same image, it suggests that the first red pixel (before demosaicing) is 57 as seen below. 

[image]

Example RAW image: `sample1.dng` found ([here: google drive](https://drive.google.com/file/d/1qk6JRg0SP137MqwqbrGhUxfUiNBB_btv/view?usp=sharing)) 

I'm probably missing some basic logic here, but is it possible to obtain the same value using `rawpy`? Thanks in advance for any pointers.

kmilos, 2022-08-08:
rawpy (LibRaw that is) and RawTherapee might be providing you with different crops of the raw array in the file, so "first pixel" is a relative term here. Does `raw_image_visible` perhaps correspond better?

Run e.g. `exiftool -a -u -s -g1` on your DNG file and see the many tags related to image size, active area, and crops. Different libraries might interpret the crops differently.

thomaskleiven, 2022-08-09:
Thanks for the pointers. Didn't manage to get exactly the same histograms pr channel using `rawpy` (LibRaw) vs. RT, but as you mention, this is probably not an issue with `rawpy` as RT might do some additional cropping, etc. Happy to move the issue to discussions.