letmaik/rawpy #180: Can it support the ACES AP0 ColorSpace?

UltraBlur, 2022-08-19:
Excuse me, I am writing a simple software to build a GUI for it as my CS lesson homework. I wonder how can I output an Image of ACES Color Space. I know ACES was supported in LibRaw, "-6".  ACES AP0 Color Space is essential in my workflow which needs to render my image to HDR Profile. 
Looking forward to your reply! This is my first time sending an issue on Github.

letmaik, 2022-08-20:
Thanks for creating the issue, you're right that this is currently missing for no good reason. In fact there is #138 which also adds two more color spaces coming in the next LibRaw release. I'll add the ACES color space now and do a release as it's not clear how long it'll be until the next LibRaw release will happen.

letmaik, 2022-08-20:
A new release 0.17.2 is being built and published. Once https://github.com/letmaik/rawpy/actions/runs/2894017481 is finished you can download it from PyPI.

closed 2022-08-20 by letmaik

kmilos, 2022-08-23:
Also note that this "ACES" colour space output by LibRaw (and therefore rawpy) is D65 adapted AFAICT, and also quantized to 16 bits (which carries some potential risk for such a huge gamut).