letmaik/rawpy #187: RawPy for Raspberry Pi4

RhoLuke, 2023-01-27:
I'm trying to install rawpy library on my Rapsberry Pi4 using pip/pip3 but in all cases I get the following error:

ERROR: Could not find a version that satisfies the requirement rawpy (from versions: none)
ERROR: No matching distribution found for rawpy

This suggests me that rawpy can't run on Raspbian OS, am I wrong?

kamikaze, 2023-01-30:
> I'm trying to install rawpy library on my Rapsberry Pi4 using pip/pip3 but in all cases I get the following error:
>
> ERROR: Could not find a version that satisfies the requirement rawpy (from [...]

rawpy is moving slow towards arm64 if at all. try finding another solution. hint: imageio will not help due to its abandoned freeimage plugin doesn't support arm64 either

letmaik, 2023-02-02:
Not sure what the exact issue is but arm64 wheels for Linux are available. It may be that you're using the 32 bit OS on raspberry.

kamikaze, 2023-02-02:
> Not sure what the exact issue is but arm64 wheels for Linux are available. It may be that you're using the 32 bit OS on raspberry. 

Sorry, my bad. Just jealous, that same arm64 wheel is not available for Apple M1/M2

letmaik, 2023-04-23:
Closing as arm64 wheels for Linux are available. If Raspberry Pi needs custom wheels, then this is not something that would be provided via PyPI. A manual build from source in that case is the best option.

closed 2023-04-23 by letmaik