letmaik/rawpy #193: postprocess to get image similar to in-camera jpeg

kamikaze, 2023-02-28:
Hi. Is there any way to post-process RAW data to get something similar to in-camera JPEG? Any params? Or any way to apply camera dcp profile? Thanks

kamikaze, 2023-03-06:
I guess these params could help a little bit:

From: https://www.libraw.org/docs/API-datastruct-eng.html
```
char* output_profile;
    dcraw keys: -o filename
    Path to output profile ICC file (used only if LibRaw compiled with LCMS support)

char* camera_profile;
    dcraw keys: -p file
    Path to input (camera) profile ICC file (or 'embed' for embedded profile). Used only if LCMS support compiled in.
```

letmaik, 2023-04-23:
Are you able to achieve a reasonable result by using dcraw directly, e.g. from https://www.libraw.org/download?

closed 2023-10-01 by letmaik