letmaik/rawpy #195: flickering when converting a sequence of DNGs using no_auto_bright=True,

ahemberger, 2023-03-03:
Hi there;

Like it says on the tin: I have a sequence of DNG images (filmed on a Canon 5dmkiii, using Magic Lantern, converted to DNG using RAWMagic). When viewing the DNG sequence on its own, my frames look as expected. I'm trying to use rawpy to convert these images (to ACES EXRs, with the help of the colour-science library). I notice that when I use the no_auto_bright=True, flag, my images 'flicker' (the brightness changes frame-to-frame). When I do not use this flag, the images convert and exhibit no flicker. 

Would anyone happen to know why this might be?

Thank you!

ahemberger, 2023-03-03:
Just discovered a thread on libraw forums indicating that no_auto_brightness can be coupled with `adjust_maximum_thr=0.0,` which seems to alleviate the flickering. 

There's also a no_auto_scale param, which also fixes the flicker, but seems to return dramatically different colors. It's not yet clear to me why one might use this option (I suspect it has to do with normalizing RGB values after debayering, to account for the higher green pixel count...the image when using it is very green).

pmisson, 2023-03-24:
The problem is probably related to the issue that I have posted. rawpy is equivalent to dcraw -4 and that makes an automatic stretching. Also, can happen because of other reasons. For example time-lapses with the shutter not completely open can flicker because the shutters do not stay the all the time exact same position when close and open between images.

letmaik, 2023-04-23:
Closing this as it is not a rawpy issue. If a specific postprocessing parameter is needed that isn't exposed yet from libraw, please open a new issue.

closed 2023-04-23 by letmaik