letmaik/rawpy #204: AttributeError: module 'rawpy' has no attribute '_rawpy'

cverrr, 2023-05-15:
Traceback (most recent call last):
  File "E:/rawpy-main/test1.py", line 1, in <module>
    import rawpy
  File "E:\rawpy-main\rawpy\__init__.py", line 8, in <module>
    globals().update({k:v for k,v in rawpy._rawpy.__dict__.items() if not k.startswith('_')})
AttributeError: module 'rawpy' has no attribute '_rawpy'

rawpy == 0.18.1
window11

letmaik, 2023-05-15:
This happens when you're cloning the repo and running Python from the repo root folder and you have `.` (current folder) in `PYTHONPATH`. Solution is to run scripts from a different folder.

closed 2023-05-15 by letmaik

cverrr, 2023-05-16:
It does not work.
Traceback (most recent call last):
  File "E:/rawpy-main/mytest/test1.py", line 1, in <module>
    import rawpy
  File "E:\rawpy-main\rawpy\__init__.py", line 8, in <module>
    globals().update({k:v for k,v in rawpy._rawpy.__dict__.items() if not k.startswith('_')})
AttributeError: module 'rawpy' has no attribute '_rawpy'