letmaik/rawpy #205: Repair bad pixels doesn't work if "cv2" isn't installed

bmilde, 2023-05-18:
I'm getting this without having opencv-python installed:

Traceback (most recent call last):
  File "convert_images3.py", line 22, in convert_image
    bad_pixels = rawpy.enhance.find_bad_pixels([image_path])
  File "/usr/local/lib/python3.10/dist-packages/rawpy/enhance.py", line 80, in find_bad_pixels
    coords.extend(_find_bad_pixel_candidates(raw, isCandidate))
  File "/usr/local/lib/python3.10/dist-packages/rawpy/enhance.py", line 112, in _find_bad_pixel_candidates
    coords = _find_bad_pixel_candidates_bayer2x2(raw, isCandidateFn)
  File "/usr/local/lib/python3.10/dist-packages/rawpy/enhance.py", line 178, in _find_bad_pixel_candidates_bayer2x2
    med = median_(rawslice)
TypeError: median() got an unexpected keyword argument 'footprint'

The error is resolved when https://pypi.org/project/opencv-python/ is installed (pip install opencv-python), but if the code is supposed to run without it as well, then this should be fixed.

letmaik, 2023-05-18:
The change to `footprint` was done recently in https://github.com/letmaik/rawpy/pull/201. Which version of scikit-image are you using?

bmilde, 2023-05-18:
I have 0.18.3

pip3 show scikit-image

Name: scikit-image
Version: 0.18.3
Summary: Image processing in Python
Home-page: https://scikit-image.org
Author:
Author-email:
License: Modified BSD
Location: /usr/lib/python3/dist-packages
Requires:
Required-by:

letmaik, 2023-05-18:
The `footprint` keyword was introduced in scikit-image 0.19 (released Dec 2021) where `selem` got deprecated and finally removed in 0.20. You will have to either use the previous rawpy version or update your scitkit-image version to 0.19 or higher.

closed 2023-05-18 by letmaik