letmaik/rawpy #206: Issues with X-H2 RAW files.

DuncanM-XR, 2023-06-26:
Think this could just be an issue with libraw versioning, but not sure. 

When processing some RAW files from the relatively new X-H2 I get a black bar on the far right.

[image]

Code for generating: 
```
import rawpy
import cv2
if __name__ == '__main__':

    # Specify the path to the RAW file
    raw_file_path = "DSCF3764.RAF"

    # Open the RAW file using rawpy
    with rawpy.imread(raw_file_path) as raw:
        # Obtain the RGB image data
        rgb = raw.postprocess()

    # Display the image using OpenCV
    cv2.imshow('RAW Image', cv2.resize(rgb, (0,0), fx=0.1, fy=0.1) )
    cv2.waitKey(0)
    cv2.destroyAllWindows()
```

Originally I assumed this was a problem with libraw as I had the same problem in darktable. But after upgrading to darktable (and therefore a more up to date version of libraw) the problem went away. 

I think I may be missing something with where to confirm the support.

letmaik, 2023-10-01:
Which rawpy version are you using? What's the libraw version in darktable?

DuncanM-XR, 2023-10-02:
I was using Darktable [version 4.2.1](https://github.com/darktable-org/darktable/releases/tag/release-4.2.1) so LibRaw 0.21.1.

For the python code snippet, I've run that on rawpy 0.16.0 (current one used in my project), 0.17.3 and 0.18.1, all producing the same result.

letmaik, 2023-10-02:
The libraw version is 0.21.1 with 0.18.1, so it shouldn't be that.
Can you share a sample file?

DuncanM-XR, 2023-10-03:
I've a put a `raf` in this google drive [folder](https://drive.google.com/drive/folders/1z8NP-PTh6LuVaWe9LRB82VHtf7uwGUBV?usp=sharing), gives me this when run with the code snippet on those versions: 
[image]

letmaik, 2023-10-15:
This is a white balance issue. See the parameters you can use here: https://letmaik.github.io/rawpy/api/rawpy.Params.html

In your case, simply using the white balance correction stored by the camera does the trick:

```py
rgb = raw.postprocess(use_camera_wb=True)
```

DuncanM-XR, 2023-10-16:
The WB is not the issue, the blackbar is. If I rerun the above code with the `use_camera_wb=True` I still get the black bar:
[image]

letmaik, 2023-10-16:
Apologies, you're right of course.
What's interesting is that the visible area is already smaller than the full area:
```py
        print(raw.raw_image.shape)
        # (5196, 7872)
        print(raw.raw_image_visible.shape)
        # (5184, 7752)
```

My first thought was that this would be the same. For some reason, libraw doesn't determine the dimensions of the visible area properly leading to the extra black bar.

Running exiftool on the RAW file gives this:

```
Raw Image Full Size             : 7872x5196
Raw Image Crop Top Left         : 21 12
Raw Image Cropped Size          : 7728x5152
Raw Image Width                 : 7752
Raw Image Height                : 5184
```

libraw's size visible size matches `Raw Image Width`/`Raw Image Height` but not `Raw Image Cropped Size`. I'm not fully sure why there are three sizes, but it must be related to that.

It would be good to check if you can get the right crop using dcraw.

letmaik, 2023-10-16:
@kmilos Do you have any idea what's going on here? I'm not that familiar with darktable but I saw it uses both libraw and rawspeed, could that explain the difference in crop behavior? When is darktable using which library?

letmaik, 2023-10-16:
Thanks @kmilos. 

@DuncanM-XR It looks like it's a bug in libraw, can you report the issue there and post the link to it here for tracking?

DuncanM-XR, 2023-10-18:
Looks like the people at libraw seem to think it is just because the X-H2 is only partially supported right now and not full supoorted. Gonna close this issue.

closed 2023-10-18 by DuncanM-XR