letmaik/rawpy #208: No source on pipy

cunyap, 2023-08-06:
Hi,

Really appreciate your tool! Could you publish the source (`tar.gz`) on https://pypi.org/project/rawpy/#files? It would facilitate the automatic cross-compilation for embedded systems.

Thanks!

letmaik, 2023-08-06:
You should be able to achieve what you want with `pip install git+https://github.com/letmaik/rawpy`? 

I generally don't publish source packages anymore for any of my binary Python packages as those more often than not lead to failed builds with obscure error messages and more people complaining and I don't have the time to handle the support requests. I treat building from source as an advanced use case that shouldn't happen by accident (pip automatically falls back to source packages if there are no binary wheels for the target Python version and/or platform, which is unfortunate).

closed 2023-08-06 by letmaik

cunyap, 2023-08-14:
Hi, 

No. I am using for example buildroot to build custom pre-compiled Linux systems for embedded systems. The default is to download the source from pypi (https://buildroot.org/downloads/manual/adding-packages-python.txt). Most package developers also upload the source and therefore facilitate the usage of such automated build systems.
I do understand that you do not have the time to help troubleshoot obscure errors. However, it would be nice to support advanced users in using your nice tool by simply publishing the source code also on pypi.