letmaik/rawpy #210: “Image too big” thrown when file does not exist

Wrzlprmft, 2023-11-16:
When I enter a wrong filename, I do not get the expected `FileNotFoundError`, but: “Image too big for processing”:

```Python
import rawpy
from pytest import raises

filename = "wrzlprmft.CR2"

with raises(FileNotFoundError):
	open(filename,"rb")

with raises(FileNotFoundError):
	rawpy.imread("non_existing_file.CR2")
```

This happens to me with RawPy version 0.18.1 and `rawpy.libraw_version` being `(0,21,1)`.

letmaik, 2023-11-16:
The error is coming directly from libraw, looks like a bug. Can you report it over there?

Wrzlprmft, 2023-11-16:
Generally yes, but I would have either only be able to relaying everything to this issue here without being able or I would have to spend some time on familiarising myself with LibRaw itself.

jtomori, 2025-02-18:
Just tested rawpy 0.24.0 with the following:
~~~ python
with rawpy.imread("non_existing_file.CR2") as 
    pass
~~~
Which has raised `rawpy._rawpy.LibRawIOError: b'Input/output error'`

It looks like it's been fixed in the meantime.

Wrzlprmft, 2025-02-18:
I cannot confirm this (with the same version of RawPy), which suggests that it depends on the operating system or version of libraw.

Wrzlprmft, 2025-02-18:
Okay, I just found something interesting. With pure Python, I get pretty much the same as you do. With IPython, however, I get:

```
LibRawTooBigError                  Traceback (most recent call last)
File rawpy_bug.py:10
      7 	open(filename,"rb")
      9 with raises(FileNotFoundError):
---> 10 	rawpy.imread("non_existing_file.CR2")

File ~/.local/lib/python3.13/site-packages/rawpy/__init__.py:20, in imread(pathOrFile)
     18     d.open_buffer(pathOrFile)
     19 else:
---> 20     d.open_file(pathOrFile)
     21 return d

File rawpy/_rawpy.pyx:413, in rawpy._rawpy.RawPy.open_file()

File rawpy/_rawpy.pyx:955, in rawpy._rawpy.RawPy.handle_error()

LibRawTooBigError: b'Image too big for processing'
```

jtomori, 2025-02-18:
Interesting, I get the same `LibRawIOError: b'Input/output error'` even when running the sample above in IPython REPL / running script with ipython (IPython 8.27.0).

Wrzlprmft, 2025-02-19:
> What OS are you using?

I am on Arch Linux. But I have also seen this issue on Ubuntu.

pierotofy, 2025-02-25:
Can confirm, I've also seen `Image too big for processing` when the file does not exist on Arch Linux. Mm.

letmaik, 2025-03-01:
@pierotofy can you reliably reproduce it? Would be good to write a unit test that reproduces it in CI.

pierotofy, 2025-03-02:
I'm going to guess this is a problem upstream with libraw; `INT64 _fsize;` is never initialized if the file does not exist https://github.com/LibRaw/LibRaw/blob/master/libraw/libraw_datastream.h#L142 https://github.com/LibRaw/LibRaw/blob/master/src/libraw_datastream.cpp#L161-L186 and some compilers probably initialize that value to `0`, but that's not guaranteed and is undefined behavior, so in some builds the value is probably some really large random number and that triggers the file size check later, which returns `LIBRAW_TOO_BIG`.

jtomori, 2025-03-02:
It might be worth reporting it to https://www.libraw.org/forum/15. They seem to be pretty quick with replying and fixing issues.

pierotofy, 2025-03-02:
I'm just not 100% sure, after looking more carefully, `open_file` has a `stat` call which *should* fail and return the correct error code if the file does not exist. https://github.com/LibRaw/LibRaw/blob/master/src/utils/open.cpp#L32

itzzdev09, 2026-08-30:
@letmaik re: a reliably reproducible case — the symptom has **changed** on current versions, which seems worth recording before anyone writes that CI test against the old message.

On `rawpy 0.27.0` / `libraw 0.22.1` (win32), the missing-file case no longer raises `LibRawTooBigError`. It raises `LibRawIOError`:

```
missing file                       -> LibRawIOError: b'Input/output error'
missing file (no extension)        -> LibRawIOError: b'Input/output error'
existing non-RAW file              -> LibRawIOError: b'Input/output error'
existing empty file                -> LibRawIOError: b'Input/output error'
a directory                        -> LibRawIOError: b'Input/output error'
builtin open() on missing          -> FileNotFoundError: [Errno 2] No such file or directory
```

So "Image too big for processing" appears to have been fixed upstream somewhere between 0.21.1 (the version in the original report) and 0.22.1 — a test asserting that message would not reproduce today.

The original complaint still stands, though, and arguably in a worse form: **all five failure modes are now indistinguishable**. A caller can't tell a typo'd path from a genuine I/O error, an unsupported format, or a directory — they all surface as `Input/output error`. Previously at least the message was wrong in a *noticeable* way; now it's uniformly uninformative.

That also means the earlier suggestion of reporting it to the LibRaw forum may be a partial fix at best. Even if libraw returned a distinct code for ENOENT, everything else in that list would still collapse together.

A `stat` on the Python side in `imread` would make the common case behave the way `open()` does:

```python
if isinstance(pathOrFile, str):
    if not os.path.isfile(pathOrFile):
        # raises FileNotFoundError, IsADirectoryError, or PermissionError as appropriate
        open(pathOrFile, "rb").close()
    d.open_file(pathOrFile)
```

Deferring to `open()` rather than raising manually keeps the errno, message and exception type consistent with the stdlib, and covers the directory and permission cases for free. It wouldn't help callers using `RawPy().open_file()` directly, so if you'd rather the check live deeper that's a different change.

Happy to send a PR with that plus a CI test, if you want it at the `imread` layer — or just the failing test on its own if you'd prefer to keep pushing on the libraw side. Let me know which is useful and I'll put it together.

---

Disclosure: investigated with AI assistance (Claude Code). The table above is actual output from running each case, not inferred.

closed 2026-08-30 by letmaik