letmaik/rawpy #227: Include Libraw snapshot releases?

lufrikas, 2024-05-11:
Hi, just wanted to ask if it would be possible to also offer rawpy-builds against the snapshot releases of Libraw? With the stable versions only being updated in a yearly interval, including the snapshots would greatly improve support for new cameras. For example, I am urgently looking for support for the Lumix S5MII, which still isn't supported in Libraw 0.21.2, but is available in the newest snapshot release.

letmaik, 2024-05-12:
See https://github.com/letmaik/rawpy/issues/106.
In principle yes, no problem. In practice, there's not much demand for it. Feel free though to submit a PR to the `prerelease` branch like it was done in https://github.com/letmaik/rawpy/pull/107/files. Then the only thing on my end would be to merge it and create a git tag.

letmaik, 2024-07-06:
Closing this. If you need it, please submit a PR.

closed 2024-07-06 by letmaik