letmaik/rawpy #233: Rawpy support for MacOS and python version 3.8 and 3.9

FirefoxMetzger, 2024-06-21:
I've noticed that rawpy doesn't support Python 3.8 and Python 3.9 on MacOS (ARM chipset). I was wondering if there is an explicit reason for this or if support for this could be added to the build matrix :)

letmaik, 2024-06-21:
No reason. Python 3.8 is soon EOL, but I'm ok adding an arm variant for 3.9. Any reason you cannot update to >=3.10?

FirefoxMetzger, 2024-06-22:
> Any reason you cannot update to >=3.10?

Personally? None. I am defaulting to 3.11.

For ImageIO I'm hoping that rawpy can eventually become our default plugin when reading raw formats instead of FreeImage which is the current default. This would allow us to phase out FreeImage which hasn't been actively maintained for years. One requirement for the swap is that ImageIO+Rawpy works where ImageIO+FreeImage works today; hence, the question if you could add builds for 3.9 and 3.8.

That said, letting 3.8 phase-out is perfectly fine since there are only a few months left before we drop it, too.

letmaik, 2024-06-22:
numpy 2 dropped support for Python 3.8, so I will also have to drop 3.8. The idea would be to add a mac arm64 wheel for 3.9 and update to numpy 2. Does that make sense for your use case?

FirefoxMetzger, 2024-06-22:
yep, that works :)

letmaik, 2024-07-06:
This is done now, closing.

closed 2024-07-06 by letmaik